TypQxQ/Sigenergy-Home-Assistant-Integration #11: Wrong consumed Power in APP and Addon

spaxxilein, 2024-09-23:
Did you notice that the correct consumption is not calculated via APP / Modbus? At least for me the inverter draws around ~ 100W 24/7 - even tough the battery is empty at night and there is no PV. You can see the difference from your sensor.sigen_consumed_power and sensor.sigen_grid_sensor_import_power. I also double checked my findings with my Shelly Pro3EM and it shows the same.

I already informed Sigenergy that this power consumption seems too high and that the consumption from the inverter is not covered in the graphs inside the APP.

Maybe you can calculate the correct consumed power before Sigenergy fixes this ;)

TypQxQ, 2024-09-23:
Modbus isn't reporting a value for consumed power.
Currently this plugin calculate it like this:
sigen_plant_active_power + sigen_grid_sensor_active_power.

I did notice some inconsistancy and was thinking of changing to:
PV + Grid Import  + Battery Dischagrge - Grid Export - Battry Charge.

This should add upp everything :D

spaxxilein, 2024-09-25:
> Modbus isn't reporting a value for consumed power. Currently this plugin calculate it like this: sigen_plant_active_power + sigen_grid_sensor_active_power.
>
> I did notice some inconsistancy and was [...]

This should fix it ;)

koddistortion, 2024-09-25:
> Did you notice that the correct consumption is not calculated via APP / Modbus? At least for me the inverter draws around ~ 100W 24/7 - even tough the battery is empty at night and there is no PV. You can see the difference from your sensor.sigen_consumed_power and sensor.sigen_grid_sensor_import_power. I also double checked my findings with my Shelly Pro3EM and it shows the same.
> 
> I already informed Sigenergy that this power consumption seems too high and that the consumption from the inverter is not covered in the graphs inside the APP.
> 
> Maybe you can calculate the correct consumed power before Sigenergy fixes this ;)
> 
> ![image](https://private-user-images.githubusercontent.com/3866673/369777003-7f25e2cb-4728-4370-9bbc-b11b83b020cf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcyNTM2MjEsIm5iZiI6MTcyNzI1MzMyMSwicGF0aCI6Ii8zODY2NjczLzM2OTc3NzAwMy03ZjI1ZTJjYi00NzI4LTQzNzAtOWJiYy1iMTFiODNiMDIwY2YucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDkyNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA5MjVUMDgzNTIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTIyZThiZDAyMzgxYjc2MDI0NjY3NzllMWU2YWViNDg2ODNiYjg0ZTNiMTkzZjEzMmFlZDQ3N2U3NDVjNjM5YSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.GTGGMSUj6_cl0sSkTWScbhJahO2b20uwVU26SS2lu5I)
> 
> Solar Power + Import from Grid should be consumed power - but its still missing ~ 100W which is the inverter itself.

wait a second, are you telling me that the SigEnergy inverter consumes 100W x 24hours = 2,4kWh per day during operation? That would result in 876 kWh per year?!

spaxxilein, 2024-09-25:
> wait a second, are you telling me that the SigEnergy inverter consumes 100W x 24hours = 2,4kWh per day during operation? That would result in 876 kWh per year?!

That is what i found out until now - i opened a ticket on the Sigenergy App but have not received any response. I also double checked with my Shelly Pro3EM that the statistics in the APP actually dont include the power consumption of the inverter. At night with minimum SOC and no solar the inverter draws 98W even according to the Sigenergy App Devices Tab. If i turn off the inverter the power draw is gone.

cjungde, 2024-09-25:
There was an update, at least in the iOS App:
[image]
Maybe its worth checking the drain again ?

TypQxQ, 2024-09-25:
I added a couple of sensors for this now:
"sigen_inverter_power_usage" for instant power.
"sigen_accumulated_inverter_energy_usage" For tracking usage.

This has nothing to do with the app updates.

I'm not surprised that it's drawing power, it's cool we can track it 😄

spaxxilein, 2024-09-26:
> There was an update, at least in the iOS App: [...]

This is the power drain of the iOS device ;) I received a stupid feedback from Sigenergy "The 100W consumption will be covered by the grid" - okay so energy prices seems irrelevant to Sigenergy support. Stupid...

spaxxilein, 2024-09-26:
> I added a couple of sensors for this now: "sigen_inverter_power_usage" for instant power. "sigen_accumulated_inverter_energy_usage" For tracking usage.
> 
> This has nothing to do with the app updates.
> 
> I'm not surprised that it's drawing power, it's cool we can track it 😄

I do know that there is a power draw for the inverter - but if its basically doing nothing at night i expected the power draw to be similar to other hybrid inverters not 3* the wattage (100W total)

spaxxilein, 2024-09-28:
Even if this is not the right place, if somebody wants to get rid of the 100W power draw at night from the inverter. The installer has to enable energy saving mode in the system settings. This will make it unable to go off grid manually and might take some time when the grid fails, but gets rid of the 100W power draw of the inverter is at idle.

micvppl, 2024-09-28:
First thig first, KUDOS @TypQxQ for this integration.  
This are my stats of new sensor and i see that there is a power drain at night.
[image]
I have asked my instalator to check app on his side and will update asap if it was not enabled by default.
I have also found something about energy saving mode and i believe it should by design be in energy saving mode??
[image]

spaxxilein, 2024-09-28:
> First thing first, KUDOS @TypQxQ for this integration. This are my stats of new sensor and i see that there is a power drain at night. ![image](https://private-user-images.githubusercontent.com/151719258/371759411-88b1d728-8610-4042-a81c-89609e2f625c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1MjkxNzMsIm5iZiI6MTcyNzUyODg3MywicGF0aCI6Ii8xNTE3MTkyNTgvMzcxNzU5NDExLTg4YjFkNzI4LTg2MTAtNDA0Mi1hODFjLTg5NjA5ZTJmNjI1Yy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyOFQxMzA3NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lMTBlNTJmNDc0YzUzZDEzZmY3NDIyYzMyNTFhNzdiODAzMjcxZjNlYzgxM2U2MjU0MmMyNjVmMmE0NjE4NWEyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.kKkdrrJUrR1sEhDeNfvY_sxQvPpMVQteLxkq6Wmc_M4) I have asked my instalator to check app on his side and will update asap if it was not enabled by default. I have also found something about energy saving mode and i believe it should by design be in energy saving mode?? ![image](https://private-user-images.githubusercontent.com/151719258/371758680-c52a7752-28dd-44b9-9767-ddc740ca7a47.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjc1MjkxNzMsIm5iZiI6MTcyNzUyODg3MywicGF0aCI6Ii8xNTE3MTkyNTgvMzcxNzU4NjgwLWM1MmE3NzUyLTI4ZGQtNDRiOS05NzY3LWRkYzc0MGNhN2E0Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTI4JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyOFQxMzA3NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05Yzk4MWI5NmU5ODM1YWRiYmUyZWZkMzkzNTYwYTQzNDZmZDlhNDQxZTQzYWMyZDlhYmQ0NmVlNmU2ZDJhM2QyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.GKZ2el1GuDAD0Y_00uc0QZ5fEQ6IiD2_0pQ5E54V5ZE)

It might depend on if you have only a smart meter or the energy gateway. I guess if you have the energy gateway the power saving is disabled by default to enable direct switch to off grid. If your inverter shows negative kW consumption during the night you likely don't have energy saving enabled.

micvppl, 2024-09-28:
I don't have a Gateway, sorry forgot to add this :)_

micvppl, 2024-09-28:
From online support FAQ:
https://support.sigenergy.com/cn/problem-details?noticeId=297&lang=en
> What are Energy Saving Mode and Performance Mode?

> 1. Energy Saving Mode: In this mode, when the load power is low, the inverter will go into standby and the battery will go into dormancy. The energy saving mode reduces the power consumption when the device is not working at night. However, there is a delay in waking up from standby when the device is in Energy Saver mode, which may affect the responsiveness of supplying power to the load.

> 2. Energy Saving Mode Load Threshold: The device will enter Energy Saving Mode when the load power is lower than this value, and the default value is 0.5% of the maximum power of the inverter.

> 3. Performance Mode: Ensures maximum self-consumption and fast response to third-party dispatch (e.g. VPP) or TOU mode in case of grid connection. However, the standby power consumption of the device will be relatively higher.

TypQxQ, 2024-09-29:
I guess this is solved in the integration. Unfortunately I can't do anything about the hardware ;D

closed 2024-09-29 by TypQxQ

spaxxilein, 2024-10-13:
> I guess this is solved in the integration. Unfortunately I can't do anything about the hardware ;D

Dont you think it would be the best option to change sensor.sigen_consumed_power to PV + Grid Import + Battery Dischagrge - Grid Export - Battry Charge as you mentioned earlier - currently the Cosumped power still lacks the inverter power draw.

TypQxQ, 2024-10-13:
> Dont you think it would be the best option to change sensor.sigen_consumed_power to PV + Grid Import + Battery Dischagrge - Grid Export - Battry Charge as you mentioned earlier - currently the [...]

I did that about 3 weeks ago 😄 :
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/blob/c040e124c762cf5bce92931608efcdf15194c8ea/modbus_sigenergy.yaml#L2260