TypQxQ/Sigenergy-Home-Assistant-Integration #29: Input data for entity "PV Generating" in dashboard view "Overview"

JuergenOrth, 2024-10-28:
I think the Binary Sensor "Sigen PV generating" should be derived from the sensor "sensor.sigen_plant_pv_power" (Modbus 30035) and not from "sensor.sigen_pv_power" (Modbus 31035). In any case, I have changed it accordingly in my setup:

```
root@pi51:~/docker/ha/config/packages# diff /tmp/modbus_sigenergy.yaml .
2329,2330c2329,2330
<           not is_state('sensor.sigen_pv_power', 'unavailable')
<           and not is_state('sensor.sigen_pv_power', 'unavailable')
---
>           not is_state('sensor.sigen_plant_pv_power', 'unavailable')
>           and not is_state('sensor.sigen_plant_pv_power', 'unavailable')
2333c2333
<           {% if states('sensor.sigen_pv_power')|round(0,"ceil") > 0 %}
---
>           {% if states('sensor.sigen_plant_pv_power')|round(0,"ceil") > 0 %}
```

TypQxQ, 2024-10-29:
Thank you. For most users this two show the same value but I've updated it now :)

closed 2024-10-29 by TypQxQ

JuergenOrth, 2024-10-29:
Thanks.
In my environment, both values are different (0.2-0.3 kW). I assume that the "Plant PV Power" is the sum of the inverter yields minus the inverter's own consumption. I only have one inverter and the results look like this: 

`$ ssh pipv bin/readmodbusall | egrep "30035|31035"
30035      7.153 kW    Photovoltaic power
31035      7.410 kW    PV power
`