TypQxQ/Sigenergy-Home-Assistant-Integration #31: Total export jumps when discharge begins

TypQxQ, 2024-11-02:
Total (and daily) export jumps when discharge begins. 

[image]

micvppl, 2024-11-07:
Did You found out why ? Have the same here. is it HomeAssistant bug?
[image]

micvppl, 2024-11-07:
[image]

TypQxQ, 2024-11-07:
Hi!
There is a new Sigenergy firmware update that came out 2 days ago ago I think. Someone reported that it fixed the issue so I am waiting for my installer to push the button (and a couple days of data) before I am looking into this. 

Very nice graph showing the error!

micvppl, 2024-11-07:
Oh, thanks for info, i will ask my installer than if there is new firmware. I have V100R001C21SPC104B065H right now.

TypQxQ, 2024-11-07:
I made a change in modbus_sigenergy.yaml file in this branch, take a look if it fixes it for you. I will also keep an eye, but my PV production is almost nonexistent when cloudy so you may have better luck 😄 

https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/tree/export/import-spike-fix

maand75, 2024-11-07:
I have updated and will check how it work against newest software.

micvppl, 2024-11-08:
> I made a change in modbus_sigenergy.yaml file in this branch, take a look if it fixes it for you. I will also keep an eye, but my PV production is almost nonexistent when cloudy so you may have [...]

I will test it up because I'm charging also from grid, dual-zone tariff.
My firmware is still not upgraded so we can check if that fix with left method helps. 
Thanks

Wictorssons, 2024-11-08:
I updated the sigenergy yaml file yesterday but still got a >10kWh jump this morning. I have not upgraded the firmware yet. I’m on SPC104 still.

micvppl, 2024-11-08:
Yup same here, left method didn't helped
[image]

micvppl, 2024-11-08:
Ok, I have now V100R001C21SPC105B072 and it looks fixed the problem. I did a quick test enabling charging with 0 discharge and then changed to slef consumption again for couple of seconds
[image]

micvppl, 2024-11-09:
Unfortunately still same issue after longer period without discharge. It seams like it depends how long it is charging before starting to discharge again...
[image]

micvppl, 2024-11-11:
What is strange from the asem day yesterday afternoon it started to work
[image]
[image]

TypQxQ, 2024-11-14:
Have you updated HA or anything else?

micvppl, 2024-11-14:
oh good catch, seems like i did updated HA later this day before afternoon. 
[image]

goatie999, 2024-12-18:
I'm not sure if it's related to the same topic but I do had my 1st real observed jump in unexpected energy export values (daily and accumulated) this morning.  The value jumped at 08:35GMT this morning by 32kWh so technically impossible for my 6kW inverter to do that in a split second :)

It's sure had an impact on the Energy Dashboard view of the day (which I have set to track the various Accumulated value sensors from this integration), where you'll also see a match off computed "Consumed Battery" figure

[image]

[image]

TypQxQ, 2025-01-12:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/issues/45#issuecomment-2585668824

TypQxQ, 2025-01-12:
I made a new test branch if you would like to try it out:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/tree/Trapezoidal_Totals

closed 2025-01-19 by TypQxQ