TypQxQ/Sigenergy-Home-Assistant-Integration #35: Max discharge limit "stuck"

jyourstone, 2024-11-14:
I recently installed SigenStor and found this great integration, thank you!
I've been playing around with it for a bit and I love it!

I seem to have find an issue though. When I set the value of the input text "set_sigen_ess_max_discharging_limit", the sensor "sigen_ess_max_discharging_limit" does update to the correct value that I set it to. But the battery discharge doesn't actually change.

Even more strange, I was able to change it once. **Here's the steps I took:**

1. First tried to set "set_sigen_ess_max_discharging_limit" to 0.1 while "set_sigen_remote_ems_control_mode" was set to "Maximum self-consumption (Default)". This didn't work and then I saw the text that "PCS Remote Control" must be activated.
2. I enabled "PCS Remote Control". "set_sigen_ess_max_discharging_limit" was applied and I could see that the battery discharge changed to 0.1 kW.
3. I tried changing "set_sigen_ess_max_discharging_limit" again, and while "sigen_ess_max_discharging_limit" does update, the battery does not discharge more than 0.1 kW now.

**Here's what I have tried to do to fix it so far:**

- Changing "Set Sigen Remote EMS" to "Disabled" and then back to "Enabled".
- Changing "Set Sigen Remote EMS control mode" to anything else and then back to "PCS remote control"
- Restarting Home Assistant
- Tried changing "set_sigen_ess_max_discharging_limit" multiple times.
- Making sure the battery has enough charge (duh)

While changing the remote EMS does change how much the battery is actually discharging, every time I change back to remote control, it's stuck at 0.1 kW discharging.

I can't see anything in the logs related to Sigenstor either.

Here's two screenshots that show the values:

[image]

[image]

Sigen Type: SigenStor EC 12.0 TP
Running Sigen firmware version: V100R001C21SPC105B072

Appreciate any help I can get with this.

TypQxQ, 2024-11-14:
This is verry odd!
The "Applied ESS max discharge limit" is read from modbus, so the setting is set and should not cause a problem.
This sounds like a bug in the Sigenergy firmware.

You can try:
Enable Remote EMS to "Enabled (Home Assistant control)" and "Command discharging (output power from the battery first)".
Try setting Max discharge to -1 and then to 5 which is higher than what you have available.

Really hope this helps.

TypQxQ, 2024-11-14:
If that doesn't work, try changing locale and language to US of the home assistant user.
Shouldn't matter but I haven't tested it with Swedish decimal (,) like I see in your images.

jyourstone, 2024-11-14:
Thank you for the tips!

I can't actually set max discharge to a negative value:

[image]

I did try to change to "Command discharging (output power from the battery first)" and that made the battery discharge max power. But when changing back to PCS remote control it yet again discharges 0.1 kW.

I tried changing the locale as well without any luck.

I restarted SigenStor as well, same problem.

Maybe you're right about a firmware issue in Sigenstor. I'll try to reach out to their support.

TypQxQ, 2024-11-14:
I believe "PCS remote control" is to be used when using Checkwatt or similar and then it will control the actual charge and discharge through the comport interface but I might be wrong.I used "Maximumself-consumption", "Commandcharging" or "Commanddischarging" for testing.

jyourstone, 2024-11-14:
Interesting, I didn't understand that PCS was to be used when using Checkwatt. 

I tested some more and here's what happens if I change EMS control mode:

- Maximum self-consumption: "Sigen ESS max discharging limit" is ignored
- Command discharging (output power from the battery first): "Sigen ESS max discharging limit" works!
- Command discharging (output power from PV first): "Sigen ESS max discharging limit" works!

So maybe I'm just mistaken and maximum self-consumption mode is meant to actually bypass the max discharging limit?

TypQxQ, 2024-11-15:
The Sigenergy documentation for Modbus protocol doesn’t specify.

micvppl, 2024-11-15:
I got this after recent update od SigeStore that ESS did not respected max CHARGE more than 1kW. No matter what i have done in HA it never wanted to charge more than 1kW.
What i did i went to SigeStore Mobile App and changet mode to Time-based Control and configured to charge only on default settings. It started charging in full Available max charging power. Thena i have changed mode to remote control in HA and it magically also started to work in HA after this.

darwinzhou7, 2024-11-21:
To regulate the charging, and discharging power of ESS or inverters power output, you have to make sure the following things:
1. In the user app, change "Operational mode" to remote ems first. In this mode, the singenstor's self-ems will be disabled, and the machine is ready for all Modbus power control orders.
2. As noted in the Modbus protocol : **_Power control related registers not explicitly mentioned in the "Comment" will take effect
only when the remote EMS control mode value is 0.** 
3. As noted in the Modbus protocol,  register:ESS max charging limit, will take effect when Remote EMS control mode is 3 or 4, namely:Command charging (consume grid power first) and Command charging (consume PV power first).
4.As noted in the Modbus protocol,  register:ESS max discharging limit will take effect when Remote EMS control mode is 5 or 6, Command discharging (output from PV first) and Command discharging (output from ESS first)
5. Since in mode command charging or command discharging, a power command may not be well performed due to SOC, ESS capacity, PV power, etc, registers are named as "limit". Meaning, you can only give a maximum limit, the machine will do its best to charge or discharge, but may, in some circumstances, not able to reach the "limit". 
hope this explains.
Viel Erfolg
Aus SG,China

jyourstone, 2024-11-21:
Thanks for the clarification @darwinzhou7. I'll go ahead and close this since it isn't an issue. I went ahead and created a pull request to clarify this a bit more in the section messages: https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/pull/37

closed 2024-11-21 by jyourstone