TypQxQ/Sigenergy-Home-Assistant-Integration #36: VSS Errors

Janne112x, 2024-11-19:
Hi everyone,

I went through the installation process  several times and I can not find the issue.
I am getting some Errors in the VSS, though I uploaded the file directly.

[image]
[image]
[image]

Janne112x, 2024-11-19:
[image]

Janne112x, 2024-11-19:
[image]

TypQxQ, 2024-11-20:
Please check your error logs. And if you can, change language to English for the user when capturing the errors. My german is quite bad. 
to see the logs, go to Settings, System, Logs.

EnterTheMaelstrom, 2024-11-20:
Hi,
ich hatte genau das gleiche Problem am Anfang und es lag daran, dass Modbus noch nicht freigeschaltet war. Sicher, dass der Installateur das gemacht hat? Und natürlich sichergestellt, dass die IP stimmt?

Hi,
I had the same issue at the beginning before modbus has been activated by the installer. Sure they did that? And are you sure the IP is the right one ?

Janne112x, 2024-11-20:
@EnterTheMaelstrom Danke, ich habe noch einmal im Router nachgeschaut und es wurde noch eine zweite IP vergeben.
Nun funktioniert es.

@TypQxQ There was a second IP provided by sigenstor, now it works.

holli73, 2024-11-28:
> The second IP might be your wallbox if you have one

hello, I have a 25kw inverter + 48kw battery and have 2 ip's as well - one IP is for the sigen gateway - and the other from the sigen inverter

best regards
holli

EnterTheMaelstrom, 2024-11-28:
Ah yes, that makes sense. I do not have a gateway :)

spaxxilein, 2025-01-07:
> > The second IP might be your wallbox if you have one
> 
> hello, I have a 25kw inverter + 48kw battery and have 2 ip's as well - one IP is for the sigen gateway - and the other from the sigen inverter
> 
> best regards holli

Hello, i also have a Gateway - but as far as the installer manual goes - the Gateway does not need to be connected to the Network - it is automatically connected via the Inverter (RJ45 between inverter and Gateway).

TypQxQ, 2025-01-07:
You can nowadays find the ip and mac in the app by going to settings, three dots on upper right, connection type, unit, ethernet. This will get you to this:

[image]

closed 2025-01-12 by TypQxQ