TypQxQ/Sigenergy-Home-Assistant-Integration #41: Config invalid, modbus_sigenergy.yaml", line 215, column 75

Overrule2146, 2024-12-11:
Hi, 
My first time trying and I think I followed the instruction.
I get stuck when I press "Check Configuration" under the Devloper tools.

Configuration errors
Error loading /config/configuration.yaml: mapping values are not allowed here
  in "/config/packages/modbus_sigenergy.yaml", line 215, column 75

Im running Core 2024.12.2, Supervisor 2024.11.4, OS 14.0, Frontend 20241127.7

Thanks for your dedication and what looks like a very good intergration!
Any help is appreciated!

Overrule2146, 2024-12-15:
Hi,

I managed to resolve my issue! Instead of downloading the file and uploading it to the correct folder, I copied the content of the file and pasted it into my version. Everything now seems to be working perfectly.

I'm not sure what caused the problem initially, but we can consider this resolved.

Thanks again for such a great integration!

closed 2024-12-15 by Overrule2146

TypQxQ, 2024-12-15:
That would have been my first suggestion. It's usually incompatibility between Windows and Linux linebreaks.