TypQxQ/Sigenergy-Home-Assistant-Integration #45: Wrong Export calculation

spaxxilein, 2025-01-07:
[image]

Sometimes i get wrong export values - i double checked that i am using the newest yaml from Github.

This issue does not happen everytime.

In fact there was no export at all during 9-10 in the morning:

[image]
[image]

Can you maybe help fixing the issue?

TypQxQ, 2025-01-07:
As you can see from the Sigen Grid Sensor Export Power, Sigenergy reported a negative spike in "Sigen Grid sensor active power".
Sigenergy is only reporting that value and if it's negtive then the integration asumes it's an export instead of import. The total is calculated by Home Assistant from that export value.

Easiest would have been if Sigenergy also exported other values like the daily export but it doesn't.

Janneck79, 2025-01-10:
I think, i have found the problem. The problem is how the integration sensor works in Standard. The problem with the way to high export numbers occors when you have a sudden spike in the load with a long periode with no export in advance. As i have understand it. The sensor only gets updatet when you have a change. So you have a zero point . Then you have no export at all for several hours. Then suddenly you have a high spike in export power. What the system does is to make a line from the zero point to the spike and takes the area below that as the sum. That means that the sensor calculates that there was a incline in power from zero to lets say 1kw over several hours instead of one second. 2 hours of zero and the a jump to 1kw is the a Export sum of 1kw instead of nearly nothing. I have tried that with helpers and with forcing a export power jump and the result is clear. I have made two helpers. One i have forced to update every second It is the same like in the Sigen app. The other one only updates when there is a change. My only problem is how to implement the max_sub_interval in  the yaml code. I am totaly new to coding.
[image]
[image]
[image]

koddistortion, 2025-01-10:
You could try top configure the max_sub_interval for the integration Sensor to 1 minute as follows:
```
  - platform: integration
    name: Sigen Accumulated Grid energy export
    unique_id: sigen_accumulated_grid_energy_export
    source: sensor.sigen_grid_sensor_export_power
    round: 2
    max_sub_interval:
        minutes: 1
```
See here for reference: 
https://community.home-assistant.io/t/integration-riemann-sum-value-only-updating-when-source-sensor-updates-2/678956/2

Janneck79, 2025-01-10:
If i want to use seconds do i need only to change minutes into seconds? I think the power sensor has an intervall of 5s then it would make sense to use the same intervall to get the most acurate results

koddistortion, 2025-01-11:
Unfortunately there is no good documentation available for the YAML data type `time`, which is to be used for setting the `max_sub_interval` according to the integration documentation.

So I looked through the source code of the integration and yes, just using `seconds` should be working.

This seems to be the line of code, where finally the value for `max_sub_interval` from the YAML gets validated:
https://github.com/home-assistant/core/blob/cdc96fdf6ff7ecb2193a9261888cf1aca79bc613/homeassistant/helpers/config_validation.py#L504

Also in the source, you can see that the internal data type used is `timedelta`:
https://docs.python.org/3/library/datetime.html#timedelta-objects

spaxxilein, 2025-01-11:
> Unfortunately there is no good documentation available for the YAML data type `time`, which is to be used for setting the `max_sub_interval` according to the integration documentation.
>
> So I looked [...]

So what do you suggest setting the max_sub_interval for the sigen_accumulated_grid_energy_export sensor? max_sub_interval: seconds: 5?

Janneck79, 2025-01-11:
Yes, and for all other acumulated and daily sensors as well

TypQxQ, 2025-01-12:
The trapezoidal method and "max_sub_interval" are new since I looked at this problem last. 
According to the [documentation](https://www.home-assistant.io/integrations/integration/) trapezoidal method is now prefered.

I made a new test branch if you would like to try it out:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/tree/Trapezoidal_Totals

Janneck79, 2025-01-12:
> The trapezoidal method and "max_sub_interval" are new since I looked at this problem last. According to the [documentation](https://www.home-assistant.io/integrations/integration/) trapezoidal [...]

Thanks, that is the same i have changed and it works without jumps. Even if i try to force a jump. Only thing i have additional changed is the scan intervall of the Sigen Grid sensor active power. I have an induction stove and that one is pulsing with the power use around every 5 seconds. So with the intervall of 5 seconds iget way to high or way to low readings. I have changed it to 1 second the lowest allowed intervall by Sigen.

closed 2025-01-19 by TypQxQ