TypQxQ/Sigenergy-Home-Assistant-Integration #48: Sigenergy Control from HA

spaxxilein, 2025-01-18:
Hello,

i also found another bug. I set the Sigen APP to PCS Remote control and enabled Home assistant control in the dashboard.

I want to have a way to deactivate the inverter totally because of the excessive power draw when idle / low sun production.

Therfor i am setting everything in the dashboard to Standby / Shutdown etc. - but the Inverter still dosnt go to standby.

If i go to the Sigenergy APP and click Power-Off - the inverter powers off as intended and the energy production / energy consumption of the inverter goes to 0W. I found that you implemented the Start / Stop of the inverter in this function, but if i check the start / stop code in HA i only get this values, no matter what i change:

[image]

This should only be 0 for Stop and 1 for start - but the sensor shows 65534 anytime the Inverter is in Power-On Mode.

      - name: Sigen Start/Stop code
        # 0: Stop
        # 1: Start
        unique_id: sigendev_start_stop_code
        device_address: !secret sigen_modbus_slave
        address: 40500
        input_type: holding
        data_type: uint16
        scan_interval: 10

Maybe somebody has an idea how to properly shut down the inverter using home assistant.

Regards,

spaxxilein

TypQxQ, 2025-01-18:
Hi!
First of all PCS mode means that the inverter is controlled by CheckEatt or similar. 
Haven't tried the shutdown but will try it later when I get home from work.

spaxxilein, 2025-01-18:
> Hi! First of all PCS mode means that the inverter is controlled by CheckEatt or similar. Haven't tried the shutdown but will try it later when I get home from work.

Can you please explain? As far as i understand if i choose PCS the control is 100% via Modbus - wether it comes from HA or any other software!?

TypQxQ, 2025-01-19:
Fixed in the new release so it now Shuts down and Starts from HA.

closed 2025-01-19 by TypQxQ