TypQxQ/Sigenergy-Home-Assistant-Integration #52: Multiple Inverters

CTM-GitHub, 2025-01-27:
Hi ... thank you for your efforts to date, is it possible to add support for multiple inverters ?

TypQxQ, 2025-01-27:
Hi!
Only custom made as long as it still uses YAML and not Python.
I am considering a rewrite, but it takes a great effort and the use is small.,

CTM-GitHub, 2025-01-27:
Thanks .. we have a use case and would be happy to contribute

TypQxQ, 2025-01-28:
I added some quick code to try if it works at all yet.
[Multiple Inverter Dev Branch](https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/tree/Multiple_Inverters_Dev)

To use, add the additional file named _second and optionally _third along with the usual modbus file.
Also add the new two lines to the secrets file as in the new secrets.yaml file.

Hope this works. I haven't tried this as I don't have more than one inverter and hav not done any frontend for it.

CTM-GitHub, 2025-01-28:
thanks Andrei ... much appreciated, I will update these files shortly - also happy to add you as a user if you would like to test our POC environment

CTM-GitHub, 2025-01-28:
OK .. so the first thing I noticed is that each Inverter is a host, we would need a provision for additional IP's

TypQxQ, 2025-01-28:
According to the (quite old) documentation I have if they are interconnected so that they talk to each other then you connect to the main inverter and the rest work as slaves to that one.

CTM-GitHub, 2025-01-28:
hmm .. we have been working with a local (AUS) energy provider and have confirmed that a unique system ID exists for a 'site' however each inverter/host is allocated a local IP address, having said that if you are OK to provide a private communication channel we would be happy to share access to test.

TypQxQ, 2025-01-29:
I'm away today n a trip until next week and can't help with that. You can email me next week at andrei@ignat.se
Kind regards, Andrei

stevenh, 2025-02-11:
I've got a restructured version of the yaml, which will allow for easier additional components, if that would help?

CTM-GitHub, 2025-02-11:
Thanks Steven .. that would be great

stevenh, 2025-02-13:
Here's a draft #62 with some work in progress.

TypQxQ, 2025-03-09:
Working on the next version, check here for naming conventions and if you have any sugestions:
https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/discussions/69#discussioncomment-12439552

CTM-GitHub, 2025-03-10:
Hi Andrei .. thank you for your efforts to date, we really appreciate your time.

Sig Energy refer to a Site ID which includes all equipment located at a single site, for example we have the following equipment associated with our Site ID

1 x 15KW 3 Phase inverter with 5 x 8KWh Batteries and 1 x 25KW DC Charger
1 x 15KW 3 Phase Inverter with 6 x 8KWh Batteries

With the suggested naming convention I understand a 'Plant' to be stack of at 1 - 6 batteries and 1 inverter ?

Could I suggest we replace 'Plant with Stack' and number them Stack 1 being the Stack configured as the Modbus Server followed by  Stack 2, Stack 3 and so on configured as Modbus Client 2,3 and so on.

Craig Menham
+61418736410

#69 (comment)<https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/discussions/69#discussioncomment-12439552>

TypQxQ, 2025-03-10:
Not exactly @CTM-GitHub.  A Stack is indeed one inverter with a number of batteries and an optional DC Charger.
But from the documentation a plant can be one or several stacks that are connected to the same smart meter.
One of the stacks will act as inverter with Device ID 1 and 247 while the other ones would get another Device ID.
All Plant sensors would be a total for the two of them.

A "stack" in the Sigenergy context indeed consists of:

- One inverter
- One or more battery modules
- Optional DC charger/accessories

## Regarding plants with multiple stacks:

A plant can have multiple stacks connected to the same smart meter
In a multi-stack setup, one inverter will be designated as the primary with Device ID 1
Additional stacks/inverters will be assigned sequential Device IDs (2, 3, etc.)
The document indicates that plant-level measurements do represent aggregated totals across all stacks in the system

/Andrei

CTM-GitHub, 2025-03-10:
yep .. agreed, so should the naming convention be based on 'Plant' representing the aggregated totals across all stacks in the system.

But the individual 'Stacks' being named Stack 1, Stack 2, Stack 3 and so on ?

deamer44, 2025-04-12:
> I added some quick code to try if it works at all yet. [Multiple Inverter Dev Branch](https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/tree/Multiple_Inverters_Dev)
>
> To use, add the [...]

I'm getting the following errors with 2 inverters:

`Configuration warnings
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'baudrate' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'bytesize' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'method' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'parity' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'port' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'stopbits' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 16: required key 'type' not provided
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 17: 'unique_id' is an invalid option for 'modbus', check: modbus->1->unique_id
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 18: 'device_address' is an invalid option for 'modbus', check: modbus->1->device_address
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 19: 'address' is an invalid option for 'modbus', check: modbus->1->address
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 20: 'input_type' is an invalid option for 'modbus', check: modbus->1->input_type
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 21: 'data_type' is an invalid option for 'modbus', check: modbus->1->data_type
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 22: 'count' is an invalid option for 'modbus', check: modbus->1->count
Invalid config for 'modbus' at packages/modbus_sigenergy_second.yaml, line 23: 'scan_interval' is an invalid option for 'modbus', check: modbus->1->scan_interval`

TypQxQ, 2025-04-16:
This will be added in the HACS integration coming soon.