TypQxQ/Sigenergy-Home-Assistant-Integration #57: PV production Accumulated value very wrong

spaxxilein, 2025-02-08:
Hello, i think the accumulated pv production sensor is somehow "wrong" - this is actually the first time i recognized this. Example: 15-16 o`clock the Sigen PV power sensor never goes above 2kW but in the same time the accumulated pv energy production sensor increaess by 3kWh.

[image]

I think the sensors are standard: sigen_pv_power:
`      - name: Sigen Plant PV power
        unique_id: sigen_pv_power
        device_address: 247
        address: 30035
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 5`

sigen accumulated pv energy production:

`  - platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_pv_power
    method: trapezoidal
    round: 2
    max_sub_interval:
      seconds: 5`

Hopefully anybody can fix this issue.

githubuser0xFFFF, 2025-02-10:
Hi, first I would like to thank you for this great Sigenergy integration into home assistant - I really like it, and you did a great job. But I also can confirm that the accumulated PV production value is wrong and does not match the value provided in the Sigen Android application,

The following screenshots that where taken at the same time in Home Assistant and in the Sigen App show this:

[image]

[image]

The solar production according to the Sigen app ist 4.36 kWh. In the graph in the Sigen app you can see, that the solar production starts at ca. 8:20 and runs till 13:20 - so that means ca. 5 hours. The last recorded PV value is 1.78 kW. That means I can estimate the integral here with 5 x 1.78 / 2 = 4.45. So the PV production value of 4.36 kWh recorded in the Sigen app seems to be correct.

The PV production value in HomeAssistant is 10.55 kWh. Also the consumed energy and the energy importet from grid seems to be different. But I think the yaml code is correct:

```yaml
- platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_pv_power
    method: trapezoidal
    round: 2
    max_sub_interval:
      seconds: 5
```

So this must be a home assistant issue?

spaxxilein, 2025-02-10:
I think i fixed it. The scan interval of the sigen_pv_power sensor dosnt work at all - it only refreshes every 10s. I changed the sensor method: trapezoidal back to left 

`  - platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_pv_power
    method: left
    round: 2
    max_sub_interval:
      seconds: 5`
commented out the scan interval for the PV Power sensor:
`      - name: Sigen Plant PV power
        unique_id: sigen_pv_power
        device_address: 247
        address: 30035
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        #scan_interval: 2`

And added an automation that polls the state via Modbus of Sigen Plant PV Power every 2 seconds - this works very well and the accumulated PV sensor is back to normal.

Automatition:
`alias: SigenPVUpdate
description: ""
triggers:
  - trigger: time_pattern
    seconds: /2
conditions: []
actions:
  - action: homeassistant.update_entity
    data:
      entity_id:
        - sensor.sigen_pv_power
mode: single
`

githubuser0xFFFF, 2025-02-11:
@spaxxilein Thank you you for this nice workaround. I just implemented it an will test it today. For other people with the same issue I link to the GitGub discussion about the limit of the scan interval to 5seconds:

[Modbus scan_interval minimal 5 seconds since core 2021.6.0 update, want faster polling of scan_interval = 1 which worked until 2021.5.x](https://github.com/home-assistant/core/issues/51404)

So the scan interval is limited to 5 seconds - strange that you see a 10 seconds update. But I do no understand how a scan interval of 2 seconds should make any difference to 5 or 10 seconds when calculating the integral. Normally the PV production does not change that fast, so also an scan interval of 10 seconds should work here. Have you tried to change only the integration mode from `trapezoidal` to `left` without adding the automation?

spaxxilein, 2025-02-11:
> [@spaxxilein](https://github.com/spaxxilein) Thank you you for this nice workaround. I just implemented it an will test it today. For other people with the same issue I link to the GitGub discussion [...]

I think it is better to have more values in order to calculate all sensors in home assistant. For example during the 10s period the value can be 5kW at the beginning of the reading and 5kW at the end of the 10s period, but in the middle its totally possible to have only 1kW of PV production for the other 8 seconds (e.g. clouds). If you have a value every 2 seconds the pv production will in theory be 5x more accurate than having a value only once every 10s.

i think the scan_interval does not work for Modbus sensors.

Janneck79, 2025-02-11:
> [@spaxxilein](https://github.com/spaxxilein) Thank you you for this nice workaround. I just implemented it an will test it today. For other people with the same issue I link to the GitGub discussion [...]

There is no Update limit anymore, i have put it on one second and it is updating every second. Only Sigen is limiting it to one second. in their specs.

githubuser0xFFFF, 2025-02-11:
@Janneck79 But according to this link, the scan interval is limited to 5 seconds. Or do you have some other source that this limit is gone?

https://github.com/home-assistant/core/issues/51404

Janneck79, 2025-02-11:
> [@Janneck79](https://github.com/Janneck79) But according to this link, the scan interval is limited to 5 seconds. Or do you have some other source that this limit is gone?
> 
> [home-assistant/core#51404](https://github.com/home-assistant/core/issues/51404)

Put the interval on 1 and view the changes. It updates every second in my case.

spaxxilein, 2025-02-12:
> Put the interval on 1 and view the changes. It updates every second in my case.

You adjusted the scan_interval of the modbus sensor and it worked? I tried changing it to 1s - rebooted HA and still only 10s updates ???

githubuser0xFFFF, 2025-02-12:
@spaxxilein @Janneck79 I can also confirm that this does not work. Even with a scan interval of 1 the sensor is updated every 10 seconds - and that's the way it is imlemented.

Janneck79, 2025-02-12:
Have updated HA yesterday and have noe the same problem with the pv integration sensor. Nearly the double like it should be. All other integration sensors seems to be ok.

rumblemumble, 2025-02-12:
> Have updated HA yesterday and have now the same problem with the pv integration sensor. Nearly the double like it should be. All other integration sensors seems to be ok.

the same for me .. also the power consumption of the unmetered devices looks weird (grey bars) .. it looks like that the production is counted as consumption even 80% of it was sent to the house battery.

[image]

Janneck79, 2025-02-12:
> > Have updated HA yesterday and have now the same problem with the pv integration sensor. Nearly the double like it should be. All other integration sensors seems to be ok.
> 
> the same for me .. also the power consumption of the unmetered devices looks weird (grey bars) .. it looks like that the production is counted as consumption even 80% of it was sent to the house battery.
> 
> <img alt="Image" width="717" src="https://private-user-images.githubusercontent.com/59135775/412606439-828e1e48-51c8-4505-b1f6-168a4ad5a755.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzkzOTY5NjQsIm5iZiI6MTczOTM5NjY2NCwicGF0aCI6Ii81OTEzNTc3NS80MTI2MDY0MzktODI4ZTFlNDgtNTFjOC00NTA1LWIxZjYtMTY4YTRhZDVhNzU1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMjEyVDIxNDQyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJhNDg1MDYxMzBiYWFkNjM1NGYwMzdkODM5ZDhhNDBmYzAwYjViMjY3NGJlYjg1YzBhY2MxNDNlNjM2NGYwMmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nmezvx2AHhSy_wjoJcbTV0pLiyBUijSgKzi3XHDFyE0">

The consumption is only the sum of pv production , grid and battery. If the pv production shows way to high values then the consumption is way to high as well. Can not find any difference in the pv integration sensor to grid and battery, but at least in my case it is the only one which shows wrong numbers. Pv power is the same like in the app. Somthing gets wrong after the HA update.

azebro, 2025-02-13:
> [@spaxxilein](https://github.com/spaxxilein) [@Janneck79](https://github.com/Janneck79) I can also confirm that this does not work. Even with a scan interval of 1 the sensor is updated every 10 seconds - and that's the way it is imlemented.

Interesting, as it works fine for me. Update every 1s. 
I have the same issue on wrong numbers though.

Janneck79, 2025-02-14:
Have changed the power sensor in the integration sensor to plant pv power and i get now the same readings like in the app. In my case nearly nothing due to snow on the panels. I like the plant pv power more anyway. It don‘t care how much the panels are producing. It is more important for me how much power i can use like the plant sensor is showing. 

[image]
[image]

azebro, 2025-02-14:
Has anybody by any chance opened an issue with HA, as it seems related to recent updates?

azebro, 2025-02-14:
> Have changed the power sensor in the integration sensor to plant pv power and i get now the same readings like in the app. In my case nearly nothing due to snow on the panels. I like the plant pv [...]

Can you share your config? I am still getting very wrong values for both PV production and Plant PV production.

se-wen, 2025-02-17:
The same issue here 
- 15,7 kWh - Sigen App for today
- [image]
- 32,95 kWh - HA Accumulated PV energy production sensor for today
- [image]

TypQxQ, 2025-02-17:
I can't replicate this. Works fine on my system. 
A request is made to Sigenergy to expose the daily and total energy. Home Assistant is relying on instant values provided by Sigenergy and there are issues with it apparently. 
Also in Sweden the values differ between Sigenergy and grid owner so there will always be differences but not quite this large.

WiesiDeluxe, 2025-02-17:
on my system exactly the same - near double in home assistant energy dashboard from mysigen app...

Wartopia, 2025-02-17:
> I can't replicate this. Works fine on my system. A request is made to Sigenergy to expose the daily and total energy. Home Assistant is relying on instant values provided by Sigenergy and there are [...]

First thank you for making this, your like my hero for this month. Thanks to your work iam building a smarthome with HA now, around your Sigen HA work. Would be nice if this would be in HACS at some point.

This has nothing to do with country or whatever. I hope Sigen will expose those, would be allot better.....

Also the AI SOC strategi and tarif page from the app would be a blessing. 

Your work with apex is nice, but i want to see the future working of the battery AI and not the past. 

Like i say. The live values are correct. Just the counting or adding goed wrong. On all stats. 

My pv system is a SolarEdge modbus kWh meter and not the Sigen inverter. I just added my modbus live kW link to your total Sigen and daily counter and somewhere there is the error in counting.

For me, its like 10kWh PV today more then in the real world. Grid was also off by like 1kWh.

Charge and discharge by like 3kWh.

Something is counted double sometimes somewhere between the 5-10 seconds refresh. When i look sometimes i see the second counter going from 1-3 and sometimes it doesnt stop at 5 or 10 seconds but at like 13 seconds.

I cant fix seem to fix it..... Whatever i try....

rumblemumble, 2025-02-17:
Today I have the correct values 24,23 in app and 26,25 via HA (the difference should come from the power lost in transforming and for Sigen itself)

I've uncommented the max sub interval

`  - platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_pv_power
    method: left
    round: 2
#    max_sub_interval:
#      seconds: 1`

but not really sure if this did the trick or the last HA updates and Updates from Sigen for the system. (there were at least two updates in the last 10 days when I'm not wrong)

Wartopia, 2025-02-17:
> Today I have the correct values 24,23 in app and 26,25 via HA (the difference should come from the power lost in transforming and for Sigen itself)
> 
> I've uncommented the max sub interval
> 
> * platform: integration
>   name: Sigen Accumulated PV energy production
>   unique_id: sigen_accumulated_pv_energy_production
>   source: sensor.sigen_pv_power
>   method: left
>   round: 2
> 
> commented out: max_sub_interval: seconds: 1
> 
> but not really sure if this did the trick or the last HA updates and/or the updates from Sigen for the system. (there were at least two updates in the last 10 days when I'm not wrong)

Mmmm i will test this. I have latest HA and Sigen updates. Yes there was multiple updates.

But just now i saw something new. Something impossible.

I have a Sigen 12K inverter and iam unloading live at 14.1 kWh which is impossible. 13.2 kWh is the max and that is also in the app. My live powercard also gets this wrong data.

When i check the 3 fases in HA, its 3x4400 which results in 13.2 which is right.

So i dont het this at al anymore. 

[image]

[image]

[image]

WiesiDeluxe, 2025-02-17:
I suspect that the error lies in the scaling factor – the accumulation sensors are currently using a scale of 0.01, which causes the values to be displayed at roughly twice the expected amount. Often, the inverter provides values for these registers that need to be multiplied by 0.005 to display the correct kWh. Does someone you have any indication in the inverter’s documentation found about which conversion factor you should use? If it indeed states 0.005, then the scaling value in the configuration would need to be adjusted accordingly.

WiesiDeluxe, 2025-02-17:
sorry guys forgot this, energy dashboard shows crazy stats after this
now i try
max_sub_interval:
    seconds: 10
on the sigen_accumulated sensor...
we see tomorrow :)

azebro, 2025-02-18:
It seems that after last updates the numbers are back to normal. Looks like HA got fixed.

Wartopia, 2025-02-18:
> It seems that after last updates the numbers are back to normal. Looks like HA got fixed.

Iam using all the latest updates and its still off vs reality.

Now iam using midpoint and its only 10% more then in reality. So a bit better, but still off.

Wartopia, 2025-02-18:
Still not working. Ill try something new.

Sigen counter: Wrong.

[image]

SolarEdge modbus counter:  Right.

[image]

WiesiDeluxe, 2025-02-18:
New day new data 😂
Today mysigen says 70,25kwh, home assistant dashboard says 106kwh, so its less difference to yesterday…but still not same…
My ideas going out…anybody idea we can test?

Wartopia, 2025-02-18:
I dont know. I guess i will replace the whole code and start over. 

The charge and decharge are spot on. Same formula. This just doesnt make any sense at all.

[image]

[image]

wimgoeman, 2025-02-25:
I am also facing a difference in the reporting of energy produced between HA and the sigen app, so I started doing some debugging.

With a Python script using the same modbus library which HA is using, I printed the values of address 30035, which according to this config represents the sigen_pv_power sensor, which in turn is used to calculate the accumulated energy production. The values printed are consistently lower than the values reported for pv_power in HA.
So maybe the issue is not the calculation of the accumulation, but somewhere in the modbus integration? Does the modbus integration do something with the values it reads from pyModbus?

Wartopia, 2025-02-25:
Iam still not sure what.

Only the pv stuff goes wrong now.

With the direct modbus readings from the SE i get good readings in the energy dashboard. But the Sigen HA readings are 0.3kWh off with the same sensor. Way less then beforehand tho with the Sigen pv sensor.

The energy dashboard uses the accumulated sensor from the SE itself. So no onsite calculations anymore. That solved it for me.

Gloomyeye, 2025-02-28:
> Iam still not sure what.
>
> Only the pv stuff goes wrong now.
>
> With the direct modbus readings from the SE i get good readings in the energy dashboard. But the Sigen HA readings are 0.3kWh off with [...]

Hello @Wartopia , could you sum up what you have done to get the right values? Which sensors do you use to get the right values in the energy dashboard?

Wartopia, 2025-02-28:
> Hello [@Wartopia](https://github.com/Wartopia) , could you sum up what you have done to get the right values? Which sensors do you use to get the right values in the energy dashboard?

Like i said, i dont use the Sigen sensor or Sigen calculations anymore. I could not get it to work.

I use the modbus sensor from my SolarEdge unit and it works.

If i use the SolarEdge sensor in the Sigen calculations its still a little off.

So all i can say. Its something in the script or and in the Sigen sensor. It has nothing to do with HA.

TypQxQ, 2025-03-01:
There are reports of that the values showed by the app does not corelate to the values from the grid provider in the Swedish Facebook group.

There is no way to have exactly the same values here as in the app because it is read at diffrent times and added together.
Home Assistant has had multiple problems with the calculations. I will try to bypass it in next version that I am working on, a HACS compatible integration.

garymc, 2025-03-01:
The Accumulated PV Energy figure for me is approx double what it should be since I updated the modbus_sigenergy.yaml code a couple of days ago.

fbradyirl, 2025-03-01:
Seeing the same issue also. More than double for PV daily.

TypQxQ, 2025-03-01:
What version of HomeAssistant are you guys seeing double PV running? Is it the latest?

Gloomyeye, 2025-03-01:
Yes, I installed the latest beta version of 2025.03. Double value is shown for me, too.

TypQxQ, 2025-03-01:
Very strange. I cannot reproduce it. Are you seeing double the instant production too?

seud0nym, 2025-03-02:
> Are you seeing double the instant production too?

Not for me. Instant production is basically the same (very overcast here right now and the app shows 1.4kW production and HA is showing 1.3kW). But the app shows 1.93kWh production so far today, versus HA showing 3.2kWh production. Battery discharge/charge is basically the same in both the app and HA.

Just as an aside: I had to alter the defiinition of `Sigen Consumed power` to use `sensor.sigen_plant_pv_power` instead of `sensor.sigen_pv_power`. I have 7 panels attached directly to the Sigenergy inverter, but I also have 13 panels with Enphase micro-inverters atached through the gateway that don't get counted if you only use `sensor.sigen_pv_power`.

ZB-42, 2025-03-02:
I also get very high accumulated PV Energy readings when I upgrade to the latest version of HA. The instantaneous power figure is perfect, it's just the accumulated energy. Works fine if I revert to HA version 2025.1.4 but with any HA 2025.2.x versions it goes wrong. I am using version 2025-01-19 of modbus_sigenergy.yaml in all cases.

Another interesting note is that when I set up my own integrator to convert from power (kW) to energy (kWh) this also shows the wrong (too high) value.  So that suggests that it's something to do with integrals in the main HA code doesn't it? 

The weird thing is that none of my other (self-defined) energy integrators show the wrong values, only the PV Power. Wierd.

garymc, 2025-03-02:
I'm on 2025.2.5.

I have a 13kwh sigenergy battery and 5.7kwh of solar panels. Standard install.

TypQxQ, 2025-03-02:
Weird problem. Someone just reported it might have something to do with the Energy dashboard. Do you use it? Can you try playing around with that?

ZB-42, 2025-03-02:
> Weird problem. Someone just reported it might have something to do with the Energy dashboard. Do you use it? Can you try playing around with that?

I do not use the energy dashboard, but I can display it. I have set up my own energy integrators and use these in Utility Meters to track all the values I'm interested in because I wanted to track the data on a daily, weekly and monthly basis. 

I have checked the Accumulated PV Energy values shown in the Energy Dashboard against those calculated by my integrators and shown in my gauges. They match exactly when running HA 2025.1.4. When I get time over the next day or two I'll update to a newer version of HA where it goes wrong and run some more comparison tests. 

The trouble is, when I do this and then revert back to the older version that works, I lose the recent history of all the values I am tracking which is a real pain. I takes a lot of time to calculate the missing data. But I'll do it in the hope it helps to narrow down the cause of this problem. 

I'll post the results when I have them.

Just for Info my system config is:
11kW        Solar PV Array (3 strings directly coupled to Sigenergy inverter)
10kW        Three Phase Siignergy  Inverter
16kwH      Battery
Location:  Australia

fbradyirl, 2025-03-02:
> What version of HomeAssistant are you guys seeing double PV running? Is it the latest?

Latest yes.

Stingra-VI, 2025-03-02:
Hi @ll,
so I am also getting wrong statics from the
"[sensor.sigen_daily_pv_energy_production](http://homeassistant.local:8123/developer-tools/state#)"

Somehow that data does not match the data at the MySigen app. I have tried multiple data points to mitigate the problem for me.

But, what I have found now, might be a indicator for the issue?

When I verify the sensor information for: "sensor.sigen_daily_pv_energy_production"
at "developer tools", "states"
I have "**59,84**" where as the 

MySigen data today:
Solar Generated: **28,59** kWh
Home Usage: 12,17 kWh
Solar "current": 3,9 kWh

ZB-42, 2025-03-02:
If there is an issue with time zone settings that could certainly explain what I am seeing. I am in a +10:00 time zone and my reading are about an additional 10 hours worth of production.

seud0nym, 2025-03-03:
> sensor.sigen_daily_pv_energy_production

This is just a Utility Meter derived from the problem accumulation sensor (`sensor.sigen_accumulated_pv_energy_production`). It is the same value, but using the value at last midnight as a zero base. It is always going to show the same discrepancies. I don't think the display timezone is going to cause an issue, as long as it actually resets at midnight.

For my 2 cents worth of contributions to fixes or work arounds: Last night I set the scan_interval of the sensors that contribute to accumulation sensors to 1 second, and changed the accumulation sensors to use the Left method, round to 3 decimal places and removed the max_usb_interval. I also have to use  `sensor.sigen_plant_pv_power` instead of `sensor.sigen_pv_power` otherwise the production from my Enphase micro-inverters doesn't get included.

Current comparison between HA and mySigen:

| Sensor| HA | mySigen |
|--------|----|-------|
|sigen_daily_pv_energy_production | 19.436kW | 19.47kW |
|sigen_daily_energy_consumption|5.028kW|4.36kW|

So, not perfect, but better than double.

Actual changes shown below (line numbers are based on the yaml version dated 2025-01-19):

Lines 48-62:
```
      - name: Sigen Grid sensor active power
        # Data collected from grid sensor at grid to system checkpoint
        # > 0: power from grid to system
        # < 0: power from system to grid
        unique_id: sigen_grid_sensor_active_power
        device_address: 247
        address: 30005
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 1
```
Lines 270-281:
```
      - name: Sigen Plant PV power
        unique_id: sigen_pv_power
        device_address: 247
        address: 30035
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 1
```
Lines 283-296:
```
      - name: Sigen Battery power
        # <0: discharging
        # >0: charging
        unique_id: sigen_battery_power
        device_address: 247
        address: 30037
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 1
```
Lines 2263-2277:
```
      - name: Sigen Consumed power
        unique_id: sigen_consumed_power
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        state: >-
          {{
            (
              states('sensor.sigen_plant_pv_power') | float(0)
              + states('sensor.sigen_battery_discharging_power') | float(0)
              + states('sensor.sigen_grid_sensor_import_power') | float(0)
              - states('sensor.sigen_battery_charging_power') | float(0)
              - states('sensor.sigen_grid_sensor_export_power') | float(0)
            ) | round(3)
          }}
```
Lines 2426-2470:
```
sensor:
  - platform: integration
    name: Sigen Accumulated energy consumption
    unique_id: sigen_accumulated_energy_consumption
    source: sensor.sigen_consumed_power
    method: left
    round: 3
    #max_sub_interval:
    #seconds: 5

  - platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_plant_pv_power
    method: left
    round: 3
    #max_sub_interval:
    #seconds: 5

  - platform: integration
    name: Sigen Accumulated Grid energy export
    unique_id: sigen_accumulated_grid_energy_export
    source: sensor.sigen_grid_sensor_export_power
    method: left
    round: 3
    #max_sub_interval:
    #seconds: 5

  - platform: integration
    name: Sigen Accumulated Grid energy import
    unique_id: sigen_accumulated_grid_energy_import
    source: sensor.sigen_grid_sensor_import_power
    method: left
    round: 3
    #max_sub_interval:
    #seconds: 5

  - platform: integration
    name: Sigen Accumulated Inverter energy usage
    unique_id: sigen_accumulated_inverter_energy_usage
    source: sensor.sigen_inverter_power_usage
    method: left
    round: 3
    #max_sub_interval:
    #seconds: 5
```

ZB-42, 2025-03-03:
@Stingra-VI 

I looked further into the time zone issue at my end and I don't think this is what is causing the high values. Although the way the last_reset entry is stated is inconsistent with the next_reset entry it is actually correct in that it does define a 24 hour period because:

2025-03-01T23:00:00.009257+00:00  =  2025-03-02T00:00:00.009257+01:00

In my case down here in AUS I get:
 
last_reset: 2025-03-02T14:00:00.006186+00:00     ( = 2025-03-03T00:00:00.006186+10:00 )
next_reset: 2025-03-04T00:00:00+10:00

So although it would be nice to see these two entries quoted with the same time zone offset I don't think it is the cause of our issue.

Stingra-VI, 2025-03-03:
@ZB-42 

Thank you for your input.
I am a little confused about your insights.

For my understanding, if I include a sponsor with a start time  set of 23:00:00 H... instead of 00:00:01 H,

I'd have the accumulated data of that previous day (23:00-23:59) + the data of today, which would result in a value of 2 days instead of one, right?

ZB-42, 2025-03-03:
Hi,
Sorry if I confused you. What I was trying to say is that in terms of time interval calculations, 23:00:00 with a time zone offset of +01:00  equates to the same time as 00:00:00 with a time zone offset of 00:00 i.e. they both represent midnight.

Stingra-VI, 2025-03-03:
@ZB-42 

Right,
But, that is the exact problem, this is how it should be.

Please check my entry.
I have stated that the dataset shows, and that is the exact opposite, isn't it?

"last_reset: 2025-03-01T23:00:00.009257+00:00
next_reset: 2025-03-03T00:00:00+01:00"

A) Start and End in the same time zone

B) I might not be a genius about programming , but why would you use a start time of 23:00 instead of 00:01?

Adding more context for this solution would increase my understanding of data collection.

C) I just could guess that this is the reason for the false data, please see my attached image.

[image]

Stingra-VI, 2025-03-03:
Also, wouldn't it be easier to use the variable *today* for the time range "00:00:01 - 23:59:59" 

Than *yesterday* 23:00:00xx - *next day* 00:00:01

Just my 50cents on that ..

ZB-42, 2025-03-03:
@Stingra-VI 

Ok on re-reading my last comment I realise it was confusing, I'll try to do better. 

It looks like whoever wrote the code has probably converted all date/time back to UTC to use in the calculations. Now the aim is to reset the counters at midnight each day in the local time zone, in your case that appears to be CET so  UTC+01:00. So when it is midnight in Europe, UTC is 23:00:00.

So your midnight can be expressed as 00:00:00+01:00
The same time can also be expressed as 23:00:00+00:00 on the previous day

Hopefully this helps.

Stingra-VI, 2025-03-03:
Hmm, okay.
I have thought of that, but in that relation the high vales doen't make any sense to me.
Thank you.

ZB-42, 2025-03-03:
I agree, the high readings for PV energy just don't make sense. 

For me, I get perfect values with HA Core  2025.1.4 but way too high with HA Core 2025.2.3 or later.

fbradyirl, 2025-03-03:
@ZB-42 Be Safe!

Stingra-VI, 2025-03-03:
@ZB-42 
Be safe!

ZB-42, 2025-03-04:
Thanks guys. At least I'll get a chance test the battery backup capability for real.

azebro, 2025-03-04:
I have managed to align the accumulated values with the app now. I have made the following modifications:
1. frequency update increased to 2s on modbus
2. rounding changed to 3 instead of 2
3. accumulation frequency 5s

```
- platform: integration
    name: Sigen Accumulated PV energy production
    unique_id: sigen_accumulated_pv_energy_production
    source: sensor.sigen_pv_power
    method: left
    round: 3
    max_sub_interval:
      seconds: 5
      
  - platform: integration
    name: Sigen Accumulated PV net energy production
    unique_id: sigen_accumulated_pv_net_energy_production
    source: sensor.sigen_plant_pv_power
    method: left
    round: 3
    max_sub_interval:
      seconds: 5
```

[32 more replies]

closed 2025-04-16 by TypQxQ