TypQxQ/Sigenergy-Home-Assistant-Integration #73: Latest SigEnergy - Modbus Protocol Documentation

Roving-Ronin, 2025-03-07:
Hi @TypQxQ ,

I couldn't see anywhere on the Wiki / Discussions that this have been raised, and searching online the latest version I could find it from 2024 and v1.7.

If you don't have it already, thought you might like a copy of the latest version v2.4 released 5th Feb 2025.

Regards.

[SigEnergy - Modbus Protocol EN - v2-4 - 2025-02-05.pdf](https://github.com/user-attachments/files/19119543/SigEnergy.-.Modbus.Protocol.EN.-.v2-4.-.2025-02-05.pdf)

BTW are you aware of any register that publishes the lifetime PV production of the inverter? Looking and nothing springs out as doing this.

TypQxQ, 2025-03-07:
Hi! 
Thank you but actually it is updated to 2.5 (2025-02-19). 
Still no fileds for any accumulations and the few new fields are not even activated in firmware yet.

closed 2025-03-07 by TypQxQ

Roving-Ronin, 2025-03-07:
Is there a discussion / wiki area where the 'latest' could be uploaded and shared by all ?