TypQxQ/Sigenergy-Home-Assistant-Integration #75: Goes to Remote EMS mode

sdujolo, 2025-03-12:
My inverter start to go to "Remote EMS mode" and efter disable the connection to HA it stays in current mode. I have selected to use the original app in HA.

TypQxQ, 2025-03-13:
I've had a report of a similar situation that was caused by network instability. HA failed to poll Modbus state from Sigenergy and guessed what mode it was in.

sdujolo, 2025-03-13:
So something is needed to be fix in HA it self not in this integration?

micheal-howard, 2025-03-24:
I have had the same issue. Its occurred three times over the past two days. Something switched the operation mode to Remote EMS.

shoog03, 2025-03-26:
I can confirm that this is a persistent issue which I have had since I got my Sigen system. 
It renders the integration virtually useless because its impacting the ability of me to use the solar in the most efficient way. 
Always seems to reset to EMS mode (7). I attempted to set the value to AI (1) in the settings but it doesn't work to override the EMS mode.

Hope this can be resolved so that I can carry on using this very useful integration.

TypQxQ, 2025-03-26:
You can try disabling the automation scripts for this. 
[image]

shoog03, 2025-03-26:
OK thanks for the heads up. I have disabled all of those and will report back if it resolves my issue.

cjbowen79, 2025-04-15:
I also bumped into this issue unfortunately.  I was part of a VPP already (Amber Smartshift) and this unfortunately removed the link to that automatically. :(

Have disabled automations as suggested now.

Would be great if this wasn't automatically adjusting the current setting.

I had planned to use the integration for dashboards to start with but not necessarily making changes.

TypQxQ, 2025-04-15:
There will be a readonly mode in the new HACS version. For now disabling it manually is a way to go. I am considering putting all readonly by default in this integration too.

closed 2025-04-16 by TypQxQ

HomerHH, 2025-04-27:
I had exactly the same issue. With deactivation of the EMS Automations it seems to be solved. 
Also the SOC was jumped to 6000% in the failure case. I dont know if it was the reason or just an symptom

PR #84 (2025-04-16): Automations safeguards added
Added many more conditions to automations that send values to Modbus.
This attempts to fix #83 but can only be sure after next firmware upgrade.
This should also fix #75 and other like instances.
commit 98601b80 (2025-04-16): Automations safeguards added

Added many more conditions to automations that send values to Modbus.
This attempts to fix #83 but can only be sure after next firmware upgrade.
This should also fix #75 and other like instances.