TypQxQ/Sigenergy-Home-Assistant-Integration #83: Upgrading firmware causing HA sending power-off to sigenstor

darwinzhou7, 2025-04-15:
Hi there, 
this is Darwin from Sigenergy, developer of the Modbus protocol of Sigenstor. 
We recently discovered that when firmware is upgraded, HA sends a power-off command to sigenstor and does not resend a power-on command, leading to many "non-working" inverters.
Can you maybe fix it in HA? 
a lot thanks

ps: We are open to collaboration in testing HA together with the community

TypQxQ, 2025-04-15:
Hi Darwin!
Thank you for reaching out!
And thank you for opening an issue here.
I think I found the problem and will relase a fix hopefully tomorow.
/Andrei

ps: How can I ask for some clarifications so the new [HACS-Sigenergy-Local-Modbus](https://github.com/TypQxQ/HACS-Sigenergy-Local-Modbus) integration  will work smoother?

Roving-Ronin, 2025-04-16:
@darwinzhou7 is that 'all' firmware upgrades or just the recent SPC108 ?

PS. Would it be an idea to create a SigEnergy account with a repositoring for Modbus (probably others  later), to which you then give your darwinzhou7 admin access to. Even if you didn't publish your actual code there, it would provide a useful place to publish the latest modbus release documents for everyone to access and also provide a place that anyone accessing SigEnergy via modbus could lodge issues (that they can then tag into the integration/repository that saw/found the issue to track it) or even lodge requests for new registers to be added?

mattgrech, 2025-04-16:
@darwinzhou7 it would be good to get some collaboration to get the modbus working for the AC charger, i have been speaking to Max Hai at Sigenergy AU but i think there is a disconnect on what is available within the firmware

closed 2025-04-16 by TypQxQ

TypQxQ, 2025-04-16:
> Hi there, this is Darwin from Sigenergy, developer of the Modbus protocol of Sigenstor. We recently discovered that when firmware is upgraded, HA sends a power-off command to sigenstor and does not [...]

This should now be fixed in the new release, but I can't test it until the next firmware update.

TypQxQ, 2025-04-16:
> [@darwinzhou7](https://github.com/darwinzhou7) it would be good to get some collaboration to get the modbus working for the AC charger, i have been speaking to Max Hai at Sigenergy AU but i think there is a disconnect on what is available within the firmware

AC Charger, DC Charger and Multiple Inverters are coming in the HACS version. It is a complete rewrite.

nygellyndley, 2025-04-16:
Hi, quick question on this.  Which version of the HA install is affected by this, is it the new HACS one or the modbus integration?

darwinzhou7, 2025-04-17:
> [@darwinzhou7](https://github.com/darwinzhou7) is that 'all' firmware upgrades or just the recent SPC108 ?
>
> PS. Would it be an idea to create a SigEnergy account with a repositoring for Modbus [...]

We will consider this. Thanks for the suggestion.

blue-anorak, 2025-04-17:
Hi Andrei, nice peace of work that you put together here. Thanks for all your efforts!

Last time I got caught by the firmware upgrade and the Sigen Device ended up in shutdown mode. 

This time I deployed your 1.0.3 release before manually upgrading the Sigen Device. It took a while and went through unknown and standby modes, but finally successfully  ended up in running mode again. 👍

TypQxQ, 2025-04-17:
I'm very happy to hear!

Roving-Ronin, 2025-04-22:
Hi @darwinzhou7 
Could you please confirm if the protocols for Sigenergy used are Sunspec compliant ?
https://sunspec.org/modbus/

PR #84 (2025-04-16): Automations safeguards added
Added many more conditions to automations that send values to Modbus.
This attempts to fix #83 but can only be sure after next firmware upgrade.
This should also fix #75 and other like instances.
commit 98601b80 (2025-04-16): Automations safeguards added

Added many more conditions to automations that send values to Modbus.
This attempts to fix #83 but can only be sure after next firmware upgrade.
This should also fix #75 and other like instances.