TypQxQ/Sigenergy-Home-Assistant-Integration #86: Sigen Plant PV Power shows 0.000 kW after update V100R001C21SPC108

TypQxQ, 2025-04-20:
### Discussed in https://github.com/TypQxQ/Sigenergy-Home-Assistant-Integration/discussions/85

<div type='discussions-op-text'>

<sup>Originally posted by **Ruviss** April 16, 2025</sup>
Subject: [Bug?] Sigen Plant PV Power shows 0.000 kW after update V100R001C21SPC108

Hi all,

Since updating to firmware version V100R001C21SPC108 on April 15th, 2025 at 15:00, the sensor.sigen_plant_pv_power entity in Home Assistant shows 0.000 kW constantly. No other system changes were made.

⚙️ System details:

Battery: Sigen 8K

PV System: SolarEdge with SMA inverter (AC-coupled)

Home Assistant: Running latest stable version

Integration: Sigenergy Home Assistant Integration by TypQxQ

📱 In the Sigen app, all values—including PV production, grid import/export—are correctly displayed and update in real time.

📉 However, in Home Assistant:

sensor.sigen_plant_pv_power = always 0.000 kW

Other sensors like battery state of charge and grid power still update

The SolarEdge values update every 15 minutes but are not suitable for dynamic automations

🧩 Suspected cause: This issue occurred directly after the firmware update, suggesting an API or communication change in the new firmware.

❓ Question: Has anyone else encountered this issue? Is there a known workaround or fix?

Any feedback is highly appreciated! Thanks again to @TypQxQ for this amazing integration.</div>

More info in Discusion #85 

Anything you can look at @darwinzhou7 ?

TypQxQ, 2025-04-20:
@RMH-7597 Please check and write here your Firmeware version and Sigenergy inverter model.

RMH-7597, 2025-04-20:
Firmware Version - V100R001C00
Device Model - SigenStore EC 6.0 SP

TypQxQ, 2025-04-20:
> Firmware Version - V100R001C00 Device Model - SigenStore EC 6.0 SP

Paste the firmware version as in HA. For example: V100R001C21SPC108B086A

Please also create a ticket with Sigen in the Sigen App.
When the Modbus address doesn't respond it's nothing I personaly can fix.

RMH-7597, 2025-04-20:
Sigen Firmware version from HA
V100R001C00SPC108B089A

I already raised it with Sigenenergy last week but not heard anything back from them so far.

Ruviss, 2025-04-21:
We did a deep-dive investigation using both QModMaster and Rilheva Modbus Poll and found something interesting around register 30035.

    Register 30035 returns 0 when queried as 32-bit signed, which used to align with PV production.

    When queried as 64-bit, it returns non-zero fluctuating values, but they match the current charge/discharge values, not actual PV generation.

This strongly suggests:

    30035 no longer corresponds to PV generation

    The HA entity sensor.sigen_plant_pv_power previously mapped to this register now gets either 0 or an invalid format

We’re now trying to identify the correct register for actual PV power output. Possibly Sigenergy changed this silently in firmware V100R001C21SPC108, either due to:

    A shift in datatype (from 32-bit to 64-bit)

    A move to a new register address

    A protocol change where PV from third-party inverters is no longer included

If anyone has documentation or updated register maps for this firmware, that would be a big help.

@TypQxQ Thanks again for escalating to a proper issue thread. We're happy to keep testing!

RMH-7597, 2025-04-21:
I had done similar before the bug was created and noticed output on 30035 using 64 bit. I have also been searching for an updated Sigenergy Modbus register but I could only find 1.7 from last year. I also suspect they have changed the register for PV and I have been slowly scanning other available codes to see if I can find anything that matches the solar output from the web app.
It does not explain though why others with solar are still getting readings. Unless they have their solar connected to the Sigen inverter and that now receives a total from the separate feeds on the inverter. Whereas feeds from other inverters detected by a wire clamp are now on a different sensor.

TypQxQ, 2025-04-21:
Thank you for your testing. This has been escalated to someone who can help at Sigenergy.
I am using a `SigenStor EC 15.0 TP` with `Firmware: V100R001C21SPC108B086A` and I have no problems with that register.

@Ruviss , Could you please add the exact inverter model and firmware you use?

Ruviss, 2025-04-21:
@TypQxQ and others,

Here are the requested details regarding my setup, which might help clarify the issue:

    Inverter: SigenStor EC 8.0 TP

    Firmware version: V100R001C21SPC108
    (Installed on April 15th, 2025 at 15:00 CET — issue started immediately after update)

System configuration:

    🔆 Solar panels: 16x 400Wp (total: 4800Wp), connected to a SolarEdge SMA inverter

    ⚡ PV is NOT connected to the Sigen inverter, but runs through the separate SMA inverter

    🔋 Battery: Sigen 8K (8 kWh)

    🖥️ Home Assistant via Modbus TCP at 192.168.178.207:502

    ✅ All other sensors (battery charging/discharging, grid import/export) continue to work as expected

❗ Issue details:

Since the firmware update, the following anomalies are observed:

    sensor.sigen_plant_pv_power remains fixed at 0.000 kW

    binary_sensor.sigen_pv_generating stays off or unavailable

    Actual PV generation is visible in the Sigen app, so the data is present but not passed to HA via Modbus

Thanks for escalating this to Sigenergy. If logs, register traces, or testing via Modbus polling tools (e.g. QModMaster or Rilheva) would help, I’m happy to contribute further.

Let me know what else you need — we’re eager to help debug this together!

hekmanViii, 2025-04-22:
Having the same prolem with:
sigen_plant_pv_power

After update it shows 0 kw in HA but show real value in sigenergy app.

I have a 3TP 12000 sigenergy

darwinzhou7, 2025-04-22:
> register address
> 
> A protocol change where PV from third-party inverters is no longer included
You actually get it right. Register 30055 is currently changed to "Self PV Power";  third-party PV power is no longer included. We will add a new register specific for third-party PV power in the upcoming version. All changes are for a better feature (a lot of new stuff is coming!) in the upcoming version
Sorry again for the confusion, and hope for your understanding. 

Best regards 
Darwin

RMH-7597, 2025-04-22:
> You actually get it right. Register 30055 is currently changed to "Self PV Power"; third-party PV power is no longer included. We will add a new register specific for third-party PV power in the [...]

Thanks for confirming what was suspected. I like possibly others who use Home Assistant to monitor battery usage and solar will find this as a significant inconvenience. As the sensor will not be available to monitor 3rd party solar until the next update. Can you give an indication when that next update will be available. If this is going to be several weeks away. How could we revert back to the previous software version so as to have the sensor back to monitor 3rd party solar.

Just as a separate query, as you are still showing 3rd party solar in the phone app. I would assume that is linked to a sensor already. Can you not advise which sensor that is or is it that all readings are not available via the Modbus interface.

Roving-Ronin, 2025-04-22:
Hi @darwinzhou7 

**No flame intended** and adding new features and extra granularity of information (i.e. PV from Sigenergy inverter vs 3rd party PV power is great !!! But from a least impact upon users and change management perspective, I would have thought that a less impactful approach would have been to:

a ) Introduce the new modbus registers that add the functionality (i.e. add dedicated register for 3rd party PV power, then:

b) Update the 30055 register to only count Sigenergy PV power (i.e. remove the 3rd party power from this existing register).

Do this would provide people (and other companies / partners that integrate with Sigenergy) the notice and time to update their code to minimise impact to users. 

PS. Could you please confirm Sigenergy Modus is Sunspec standard?

darwinzhou7, 2025-04-22:
> Thanks for confirming what was suspected. I like possibly others who use Home Assistant to monitor battery usage and solar will find this as a significant inconvenience. As the sensor will not be [...]

I am sorry, that you have to wait for the next version to release. 
Regarding sensor data, they are not available in Modbus.

darwinzhou7, 2025-04-22:
> Hi [@darwinzhou7](https://github.com/darwinzhou7)
>
> **No flame intended** and adding new features and extra granularity of information (i.e. PV from Sigenergy inverter vs 3rd party PV power is great [...]

Yes you are right. Thank you for the advice.
Sigen Modbus is not a Sunspec standard.

RMH-7597, 2025-04-22:
[@darwinzhou7](https://github.com/darwinzhou7)

I was looking to do a direct message but that option does not seem available. As I just want to comment that to be honest the response of you will need to wait for next version is not great.

> 
> I am sorry, that you have to wait for the next version to release. Regarding sensor data, they are not available in Modbus.

I have openly stated to others how good I felt this product was but to be honest the fact that changes are made which impact myself and others users. Then a response of, we will fix it in next version is far from satisfactory. Without even any indication of when that fix would be available which could be anything from weeks, months or even next year.

One of the reasons I went for this system was the ability to monitor locally without total dependence on cloud based systems which I do find a major negative aspect. As all too often cloud based systems eventually become systems you need to pay for monthly. To then have the locally connectivity have features taken away with no notice and what seem like an after thought of we will fix it in the future. Does give a very bad impression of the system as a whole. Hopefully this will be remediated soon as it is a major inconvenience for me as I am sure it is for others. As I have local monitoring of the system that allows others in my home to be able to see our energy usage and solar generation. Which helps them identify what can and cannot be done as we run off battery all day. A setup I have spent a lot of time putting in place is now effectively broken due to this one small change you made to discount 3rd party Solar from the modbus sensor.

I know what I am saying here is unlikely to make any difference to the time frame for this to be rectified. Suffice to say I think you will find your actions have done damage to your reputation and I personally would not be in a position to recommend this product to others as I have in the past.

Ruviss, 2025-04-22:
💡 Temporary workaround: Virtual PV Sensor in Home Assistant
Hi @RMH-7597 and others,

We fully understand your frustration — we're facing the same problem. While waiting for a firmware or integration update, we created a temporary solution that estimates PV generation locally using Home Assistant's template sensors.

This workaround uses:

Battery charging/discharging power

P1 power meter values (or grid import/export sensors)

A battery SOC threshold to distinguish between charging and export states

It’s not a perfect replacement, but for dashboards and automations, it works surprisingly well.

🧠 Example Template Sensor Logic
yaml
  - sensor:
      - name: "virtual_pv_generation"
        unique_id: virtual_pv_generation
        unit_of_measurement: "W"
        state_class: measurement
        device_class: power
        state: >
          {% set soc = states('sensor.sigen_energy_storage_system_soc') | float(0) %}
          {% set ch = states('sensor.sigen_battery_charging_power') | float(0) %}
          {% set dch = states('sensor.sigen_battery_discharging_power') | float(0) %}
          {% set p1 = states('sensor.p1_meter_power') | float(0) %}
          {% if soc < 98 %}
            {{ ((ch - dch) | round(0)) }}
          {% elif p1 < -100 %}
            {{ ((p1 * -1) | round(0)) }}
          {% else %}
            0
          {% endif %}

We use this in a separate YAML file (e.g., virtual_pv.yaml) and include it via configuration.yaml to avoid touching the original Modbus config.

🙋‍♂️ Suggestion for @TypQxQ and Team
If others find this useful, perhaps future versions of the integration could offer an optional virtual PV sensor, especially for setups with third-party inverters. It could be disabled by default and opt-in only — providing a fallback for monitoring and automations when native PV values are unavailable.

We’d love to hear what you and the community think about this approach. And we're happy to help test or refine it further if there's interest.

Thanks again for maintaining this great integration!

— Ruviss

maxw3l, 2025-04-23:
FWIW I was able to setup my Enphase Inverter via it's own integration to fill the PV generation gap. But it would be cleaner to when it comes back into the Sigen one.

StroppyOz, 2025-04-24:
> FWIW I was able to setup my Enphase Inverter via it's own integration to fill the PV generation gap. But it would be cleaner to when it comes back into the Sigen one.

Are you happy to share how you interfaced the two together? I've got a Fronius inverter which is linked via the Sigenstor gateway's smartport (which I assume is how your Enphase Inverter is set up too).

RMH-7597, 2025-05-22:
I saw a youtube video the other day that was about Sigenergy products coming out. They mentioned that an updated software release was coming at the end of this month. Certainly not an official statement of any kind but could be that this issue is resolved by the end of the month. Hopefully it will be fixed as I am seeing more weird things with the setup. Like the other day AI mode decided not to charge my batteries so I was down to 30% going in to peak usage period and an almost constant trickle of power to grid on AI mode when it is set to self consumption. Going back to time based control fixed it. Just mentioning in case any others looking here have experienced similar weirdness.

ArnoPW, 2025-06-12:
I am on firmware V100R001C21SPC109B093F which I think is one later than the V100R001C21SPC108. Has anyone been able to find whether Sigenergy has added a register for third party solar already?

RMH-7597, 2025-06-12:
> I am on firmware V100R001C21SPC109B093F which I think is one later than the V100R001C21SPC108. Has anyone been able to find whether Sigenergy has added a register for third party solar already?

I just applied the update to my setup and not seeing any readings on the already defined PV items. It may need an update to the modbus config in Home Assistant if it is a new sensor they have created.

RMH-7597, 2025-06-24:
I have received a copy of modbus 2.7 and have solar available again via modbus. What I did which is not the perfect fix but worked is changed. essentially changed 30035 to 30194

      - name: Sigen Plant PV power
        unique_id: sigen_pv_power
        device_address: 247
        address: 30035
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 5
to
      - name: Sigen Plant PV power
        unique_id: sigen_pv_power
        device_address: 247
        address: 30194
        input_type: input
        data_type: int32
        precision: 3
        unit_of_measurement: kW
        device_class: power
        state_class: measurement
        scale: 0.001
        scan_interval: 5

These are the listed changes for v2.7 if of any use to others:

Added plant running info registers: Third party 
inverter power(30194), Cumulative Energy
Interface(30196-30268);
Added an enumeration to register EMS working
mode(30003): “5: Full Feed-in to Grid” and “9:
Custom” Deleted hybrid inverter setting parameter: Grid
code(40501)
Appendix 1“Running state”added an enumeration:
Environmental Abnormality 0x07